Irev-Dev/cadhub #24: Make Part Card Component

Irev-Dev, 2020-10-24:
[image]
The part component will be used for displaying many parts on the parts page.
[image]
The component should accept three props with the following types:
```javascript
{
   title: string
   imageUrl: string,
   emotes: {
      emojii: string,
      count: number,
   }[] // basically an array of objects with an emojii and count properties
}
```
and example data would be:
```
{
   title: 'Ducky Key Caps',
   imageUrl: 'https://images.unsplash.com/photo-1595225386386-79c3543adbd9?ixlib=rb-1.2.1&ixid=eyJhcHBfaWQiOjEyMDd9&auto=format&fit=crop&w=1000&q=80'
   emotes: [
      {
         emojii: '❤️',
         count: 3,
      },
      {
         emojii: '😁',
         count: 2,
      },
   ]
}
```
The component should display a maxium of 5 emotes.
Bonus points for writing tests too, but not needed.

[Link to the figma design](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=12%3A449)

Later the component will also take user id as a prop and used that to fetch the gravatar, but just leave a circle there for now.

Irev-Dev, 2020-10-24:
When I say styled with Tailwind, I would expect 90+% to be done with tailwind, but it's fine to opt-out to regular CSS for situations where the CSS is very custom and not well suited to tailwind. For example the purple transparent gradient on the images.
```
background: linear-gradient(19.04deg, rgba(62, 44, 118, 0.46) 10.52%, rgba(60, 54, 107, 0) 40.02%)
```

lenadgit, 2020-10-25:
I would like to work on this task

Irev-Dev, 2020-10-25:
It's all your @lenadgit

Irev-Dev, 2020-10-26:
@lenadgit 
I just updated the setup guide in the readme,
You'll also need to run 
```yarn rw db up```
after `yarn install`
Wanted to mention it here, because you're bound to hit this issue.

Irev-Dev, 2020-11-08:
Resolved by #79

closed 2020-11-08 by Irev-Dev