Irev-Dev/cadhub #26: Style the nav bar

Irev-Dev, 2020-10-25:
Our nav-bar should look like this:
[image]
But currently looks like this:
[image]

- Styles should be done in tailwind css (note: gradients can be [like so](https://tailwindcss.com/docs/gradient-color-stops#app))
- The two fonts used are [Ropa Sans](https://fonts.google.com/specimen/Ropa+Sans?query=ropa) and [Roberto](https://fonts.google.com/specimen/Roboto?query=rob) (if they're not already in the project by the time you start this ticket please make sure they're configured with [tailwind](https://tailwindcss.com/docs/font-family#customizing))

When a user isn't signed in it should look like this:
[image]

[Link to the prisma design](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=7%3A37)

The file that needs editing is [MainLayout.js](https://github.com/Irev-Dev/cadhub/blob/main/web/src/layouts/MainLayout/MainLayout.js)

Let me know if you have any questions or need help with anything.
Ticket is flexiable and we can reduce the scope if you think any part of it is beyond your skill level.

yencolon, 2020-10-25:
Hi, I would like to help with this!

Irev-Dev, 2020-10-25:
It's all yours @yencolon

saurabh786iit, 2020-10-25:
Hey,
Please assign this to me

Irev-Dev, 2020-10-25:
Sorry @saurabh786iit, it's already assigned.

yencolon, 2020-10-26:
Hey @Irev-Dev I followed the installation steps but when running the app I got a database error 😬

> The table `dev.Part` does not exist in the current database.

Irev-Dev, 2020-10-26:
Oh hmm try `yarn rw db up`

yencolon, 2020-10-26:
Yeah, that works. Thanks.

Irev-Dev, 2020-10-26:
I just updated the read me. Thanks for catching this one @yencolon

yencolon, 2020-10-26:
Fonts should be added via CDN or local?

Irev-Dev, 2020-10-27:
Lets go local, it will be served by Netlify's CDN anyway.

AugustoZanoni, 2020-10-27:
#38 Solved

yencolon, 2020-10-27:
Hi @AugustoZanoni, thanks for collaborating, I guess @Irev-Dev can merge some our code :)

Irev-Dev, 2020-10-27:
Sorry @AugustoZanoni, I had to close your PR.

Thanks @yencolon 🚀

closed 2020-10-27 by Irev-Dev