Irev-Dev/cadhub #29: Weird looking in browser brave on phone.

boschie, 2020-10-25:
Hello,

I just opened the website on my phone but the lay-out is not looking good. 
See image:

[image]

With kind regards, 

Hendrie Bosch

Irev-Dev, 2020-10-25:
Thanks Hendrie,

The website is only in the begining of development, and not in a ready state. I'll leave this issue open for now, but I expect it to be resolved naturally as the project progresses.

Irev-Dev, 2020-10-26:
Yeah you can if want, though I think it's likely to conflict with #29.
Let me know if you still want to do it.

Matheusdoe-dev, 2020-10-26:
Yes, I want :)

Irev-Dev, 2020-10-26:
It's assigned.

Irev-Dev, 2020-10-26:
@Matheusdoe-dev 
I just updated the setup guide in the readme,
You'll also need to run 
```yarn rw db up```
after `yarn install`
Wanted to mention it here, because you're bound to hit this issue.

Matheusdoe-dev, 2020-10-26:
Alright, thanks!

closed 2020-10-28 by Irev-Dev