Irev-Dev/cadhub #32: Integrate <ImageUploader /> into User form

Irev-Dev, 2020-10-26:
Should be fairly straight forward, and I suspect will be very similar to this [diff](https://github.com/Irev-Dev/cadhub/pull/31/files#diff-c89ec35f2fb0fac5ba1c85209baac6065242092d0d72e40f4f4645a57cfd3a74) full [PR is here too](https://github.com/Irev-Dev/cadhub/pull/31)

I believe the file main file that needs to be changed is `web/src/components/UserForm/UserForm.js`

What to have a go at this one @Yash-r?

Yash-R, 2020-10-26:
Okay so what you are trying to say is we can also add image uploader at the time of signup, if yes than this can be done exactly as you have done for parts right @Irev-Dev ?

Irev-Dev, 2020-10-26:
I think it will be more that the user can edit it after they sign up, I think it would be difficult to put it into the sign up flow, since I'm using netlify's identity service, and using their modals so not in complete control of the UI for that.

The UserForm is used for both creating and editing users.

Redwood has generated users a simple CRUD operation, and I haven't woven permissions into it yet, but that doesn't matter to much for just adding the imageUploader.

Yash-R, 2020-10-26:
okay, I got it so can i try adding image uploader at the time of sign up and also editable

Irev-Dev, 2020-10-26:
Yup, but if sign up is too hard, I'll still merge your PR without it.

Irev-Dev, 2020-10-26:
@Yash-R 
I just updated the setup guide in the readme,
You'll also need to run 
```yarn rw db up```
after `yarn install`
Wanted to mention it here, because you're bound to hit this issue.

Yash-R, 2020-10-26:
Thanks, I was just about to ask this issue

Yash-R, 2020-10-27:
hey is it possible to get the URL of Netlify site @Irev-Dev ?

Irev-Dev, 2020-10-27:
If you put up a PR netlify will build a new url, if the If you put WIP in the PR title I'll know not to start reviewing it. Is that what your after?
I might need to approve it in netlify actually.

Yash-R, 2020-10-27:
sorry, I am using netlify for the first time i don't have any idea what you are talking about.

Irev-Dev, 2020-10-27:
You shouldn't have to worry about netlify.
Put a pull request up here on github. There is a netlify integration that will recognise the PR, I'll hit build and then a URL will show up within the pull request that you can access.
And if you not ready for me to have a look at the PR just put WIP (work in progress) in the PR title and I'll know not to look at it yet.

Yash-R, 2020-10-27:
okay, got i'm creating a pull request

Yash-R, 2020-10-27:
okay got it i'm creating a pull request

Yash-R, 2020-10-27:
@Irev-Dev  can you please briefly explain me what should i basically do because there are many routes and i'm little confused

Irev-Dev, 2020-10-28:
Sorry I've been slow @Yash-R .
So I've given you account admin rights.
[image]
So you should be able to use the user routes now.
If you go to http://localhost:8910/users you should see users local to your machine
[image]
You can edit those or make new ones at http://localhost:8910/users/new
The create and edit form has a plain text "Image" input
[image]
I want that replaced with the image uploader, the image uploader will still return a string, (the cloudinary id) so it shouldn't require any backend changes.
This has already been done for the parts create/edit form (http://localhost:8910/parts/new)

Does that make sense?

Yash-R, 2020-10-29:
Yes now i got it!

Yash-R, 2020-10-29:
Okay @Irev-Dev I have successfully created ImageUploader 
[image]
[image]

Yash-R, 2020-10-29:
But at http://localhost:8910/users its showing "No user yet" how can i fix this so that i can update edit rout also?
[image]

Irev-Dev, 2020-10-29:
This might be a stupid question but have you tried clicking "Create one?" if not maybe try `http://localhost:8910/users/new`. does that work?

Irev-Dev, 2020-10-29:
Oh sorry I didn't see your previous message, if you hit save, what happens?

Yash-R, 2020-10-29:
It just create a flash message "User Saved"

Yash-R, 2020-10-29:
Now its working

[image]

Yash-R, 2020-10-29:
@Irev-Dev Is it possible to show direct image rather than this link.
[image]

Irev-Dev, 2020-10-29:
Yeah I'd assume by modifying web/src/components/User/User.js

Yash-R, 2020-10-29:
@Irev-Dev  Created 
[image]

Yash-R, 2020-10-29:
What should i do next ?

Irev-Dev, 2020-10-29:
Perhaps it's time for a pull request?

closed 2020-11-01 by Irev-Dev