Irev-Dev/cadhub #33: Parts Page

Irev-Dev, 2020-10-26:
There are designs for the part page
[image]
I haven't given it enough thought to write up a good issue, but someone is more than welcome to have a crack at it.
Please use tailwind to style
[Designs can be found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=28%3A0)

rohansp, 2020-10-27:
@Irev-Dev I would like to work on this if no one is assigned.

Irev-Dev, 2020-10-27:
It's all yours. Assigned.

Irev-Dev, 2020-10-27:
Hey @rohansp, I've been thinking about this issue and I've decided it's too big and vague
> I haven't given it enough thought to write up a good issue
🤦‍♂️
So I've broken it down into the following issues instead.
#46, #48 , #49 

Let me know what you've so far and I can assign some of them to you.
This issue will still be good to have for the final layout and parts not covered by the above issues.

rohansp, 2020-10-28:
I will work on #46

Irev-Dev, 2020-11-08:
Resolved by #79

closed 2020-11-08 by Irev-Dev