Irev-Dev/cadhub #34: User page

Irev-Dev, 2020-10-26:
There are designs for the part page
[image]

I haven't given it enough thought to write up a good issue, but someone is more than welcome to have a crack at it.
Please use tailwind to style
[Designs can be found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=40%3A54)

sooster910, 2020-10-26:
Hi 👋  @Irev-Dev . Can I try to work on ?

Irev-Dev, 2020-10-26:
Go for it. Assigned.

Irev-Dev, 2020-10-26:
I just updated the setup guide in the readme,
You'll also need to run 
```yarn rw db up```
after `yarn install`
Wanted to mention it here, because you're bound to hit this issue.

MananDesai54, 2020-10-27:
hi @Irev-Dev can i work on it?

sooster910, 2020-10-27:
@Irev-Dev I might take a while to finish this up.  @MananDesai54  You can go head !

MananDesai54, 2020-10-27:
> @Irev-Dev I might take a while to finish this up.  @MananDesai54  You can go head ! 

Okay

Irev-Dev, 2020-10-27:
I decided that this issue was too big and vague, so I've instead broken it down into #45, #46 and #47.
If either of you @sooster910 and @MananDesai54 would like to work on these let me know!

For now I'm not going to assign anyone to this issue until the above 3 are done.

MananDesai54, 2020-10-28:
@Irev-Dev I'll start with #45

Irev-Dev, 2020-11-05:
resolved by #59 and #60

closed 2020-11-05 by Irev-Dev