Irev-Dev/cadhub #45: Make editable key: vaule input

Irev-Dev, 2020-10-27:
The profile have the following styled input
[image]

For this issue you need to:
- Make a react component for the above
- It should at least have 3 props 1) isEditable: boolean, 2) value: string 3) onChange: callback
- Style with tailwind css

Design can be [found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=152%3A7)
Context of where these inputs will go:
[image]

Matheusdoe-dev, 2020-10-28:
is it only to do this input editable key ? is it anything else ? I can do this.

MananDesai54, 2020-10-28:
Can I start with it ?@Irev-Dev

Irev-Dev, 2020-10-28:
Yup, all yours @MananDesai54

Irev-Dev, 2020-11-05:
Resolved mostly by #60 
https://github.com/Irev-Dev/cadhub/pull/60/files#diff-9cce7458d97c748ddccee1ed2d5aa824325f14eccc0aba00bfb60ad03d33e802

closed 2020-11-05 by Irev-Dev