Irev-Dev/cadhub #46: Make avatar, with updatable state

Irev-Dev, 2020-10-27:
The profile page will have following updatable avatar component
[image]

For this issue you need to:
- Make a react component for the above extending/modifying the [ImageUploader component](https://github.com/Irev-Dev/cadhub/blob/main/web/src/components/PartForm/ImageUploader.js)
   https://github.com/Irev-Dev/cadhub/blob/main/web/src/components/PartForm/ImageUploader.js
- It should at least have 3 props 1) isEditable: boolean, 2) imageId: string (this will be a cloudinary id used to fetch the image) 3) onChange: callback that returns the new couldinaryid when a new upload occures
- Style with tailwind css

Design can be [found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=152%3A41)
Context of where these inputs will go:
[image]

rohansp, 2020-10-28:
@Irev-Dev So, initially the user page will have the edit pencil icon on the image.User will click on this pencil and upload new image. After 100% image is uploaded the below image without the pencil icon is shown. Is this the behavior? You can assign this to me.

rohansp, 2020-10-28:
@Irev-Dev I am having trouble with yarn. yarn --force is taking too much time. Will keep you posted.

rohansp, 2020-10-28:
@Irev-Dev I am getting this error.
[image]
Need your feedback

rohansp, 2020-10-28:
@Irev-Dev yarn rw dev gives this error after opening the link http://localhost:8900 in Chrome and I see Something went wrong message on page.

rohansp, 2020-10-28:
@Irev-Dev Tried the yarn rw dev command on Windows Powershell in VS Code. Keep getting web | [HPM] Error occurred while trying to proxy request /graphql from localhost:8910 to http://[::1]:8911 (ECONNRESET) (https://nodejs.org/api/errors.html#errors_common_system_errors) 
[image]

Am I missing something here?

Irev-Dev, 2020-10-28:
Sorry I can't help you with you errors, I'm an a mac, and I haven't run into anything like this. Looks pretty system specific to me

> @Irev-Dev So, initially the user page will have the edit pencil icon on the image.User will click on this pencil and upload new image. After 100% image is uploaded the below image without the pencil icon is shown. Is this the behavior? You can assign this to me.
So whether the pencil shows or not should be controled by `isEditable` prop, that way the same component can be used to display the avatar on the profile page, and then again on the edit profile page where there pencil will show up.

Irev-Dev, 2020-10-28:
@rohansp I've had a look on both the redwood forums and the discord and couldn't find anyone who's hit the same error.
I've found the community to be very responsive if you wanted to ask on the discord. https://discord.com/invite/jjSYEQd

Sorry about the hassle, there's no obligation to keep push through it if you not enjoying it anymore.

mvarrieur, 2020-10-29:
@rohansp @Irev-Dev I found that redwood API/backend just crashes completely on the latest v12 version of Node (12.19.0). 12.18.x was working for me. Looks like you might be having the same issue. I didn't try v14 since it just went LTS yesterday

rohansp, 2020-10-29:
Okay Thanks Michael. I will try it out.

Irev-Dev, 2020-11-05:
Resolved by #59

closed 2020-11-05 by Irev-Dev

rohansp, 2020-11-06:
@Irev-Dev Sorry, couldn't find time to complete this.

Irev-Dev, 2020-11-06:
@rohansp No need to apologise. Thanks for having a go.

Did you make a start purely because of hacktoberfest, or do you like the project and want to contribute to some other part of it?

rohansp, 2020-11-09:
@Irev-Dev I started with Hacktoberfest but idea of project using React and having interesting design increased my interest in this project.

Irev-Dev, 2020-11-11:
Nice @rohansp 😄 .
I've set up a github project board https://github.com/Irev-Dev/cadhub/projects/1
It's a little bare at the moment, but I hope to fill up the backlog with things I'd like to get done, if any of them interest you at anypoint feel free pick them up or chat to me about it. I'm also open to any ideas you have for the website.

I've also setup a discord server, its bit easier to chat if you want to spit-ball ideas. 
Up to you
https://discord.gg/57J4y4AR