Irev-Dev/cadhub #47: Override "rich-markdown-editor" styles

Irev-Dev, 2020-10-27:
The profile page uses "rich-markdown-editor" for displaying and editing markdown though it needs to look like this
[image]
But currently look like this with default styles
[image]

For this issue you need to research rich-markdown-editor to find the best way to style it and to style it.
Most likely it will be the [inbuilt theming](https://github.com/outline/rich-markdown-editor#theme)
though for parts not appropiate for the theming above try and use tailwind

Design can be [found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=152%3A68)
Context of where this will go:
[image]

YosephKS, 2020-12-12:
@Irev-Dev can I work on this next?

Irev-Dev, 2020-12-12:
@YosephKS You can, though I did add some overrides already, not using the rich markdown editor themes but instead with some overriding css, so it already looks "OK" not exactly the same as the design, the headings are still black. It looks like this atm.
[image]

If you want you can style it all through the theme or more overrides.

Because it's partly done I had considered closing this issue since it's not super valuable. If you'd prefer I can write some more issues, there are things to be done that I just haven't planed out yet.

The overrides I added are in `web/src/index.css`

closed 2021-06-30 by Irev-Dev