Irev-Dev/cadhub #48: Make "user-name / part" component

Irev-Dev, 2020-10-27:
The page needs the following component
[image]

For this issue you need to:
- Make a react component for the above
- It should at least have 2 props 1) userName: string, 2) partName: string
- Style with tailwind css

Design can be [found here](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=158%3A0)
Context of where these inputs will go:
[image]

Irev-Dev, 2020-10-27:
Font is roboto and has already been added to the tailwind config.

lyonsbp, 2020-10-27:
I can take this one!

Irev-Dev, 2020-10-27:
Hey, @lyonsbp thanks for volunteering. I've assigned it to you, but please note there is overlap between this issue and #33, and we might have to negotiate with @rohansp if there are conflicts.

lyonsbp, 2020-10-27:
Gotcha, I just submitted the PR anyway since I didn't see that message.

PR #50 (2020-10-28): Add UserPart component
Resolves #48
closed 2020-10-28 by Irev-Dev