Irev-Dev/cadhub #65: Add welcome console.log

Irev-Dev, 2020-11-06:
If you open up the dev tool console on [theguardian.com](https://www.theguardian.com) there is a well formated console.log that invites folks to apply for jobs.
[image]

I'd like to have the same thing inviting folks to get involved in the project. This will make a good first issue since you don't need prior knowledge of how the project works. Just need to google how to format console logs.
I think the best way to add it is in a `useEffect` hook on the mainlayout since that's on every page.
`web/src/layouts/MainLayout/MainLayout.js`

ezekielnizamani, 2020-11-07:
Hello. I would like to work on it

Irev-Dev, 2020-11-07:
Great, thanks. I've assigned it to you.
You should work of the `dev` branch, NOT `main` as I've got a lot of changes in dev about to go into main soon.
Let me know if you need any help with setup.

ezekielnizamani, 2020-11-07:
Okay thank you

Irev-Dev, 2020-11-07:
@ezekielnizamani, Lets make the message:

## Hey, 👋

CadHub is in active development - Want to lend a hand?
[https://github.com/Irev-Dev/cadhub](https://github.com/Irev-Dev/cadhub)

ezekielnizamani, 2020-11-07:
@Irev-Dev I would love to

ezekielnizamani, 2020-11-07:
> @ezekielnizamani, Lets make the message:
> 
> ## Hey, 👋
> CadHub is in active development - Want to lend a hand?
> https://github.com/Irev-Dev/cadhub

> should  I add that in the console log?

ezekielnizamani, 2020-11-07:
@Irev-Dev  is this what I have to add to the console log?

Irev-Dev, 2020-11-07:
@ezekielnizamani Yes thanks.

ezekielnizamani, 2020-11-07:
@Irev-Dev  okay cool

ezekielnizamani, 2020-11-07:
@Irev-Dev  I use npm this my first time using yarn I am facing this problem when I run yarn install There appears to be trouble with your network connection. Retrying...

Irev-Dev, 2020-11-07:
@ezekielnizamani Oh okay, not sure I can help with that specific problem with yarn.
You should be able to use the comands using npm, try the following.

Install dependencies
```terminal
npm install
```

Initialise the db
``` terminal
npm run rw db up
```

### Fire up dev
```terminal
npm run rw dev
```

ezekielnizamani, 2020-11-07:
okay thank you

Irev-Dev, 2020-11-07:
@ezekielnizamani if it helps here's the line they use on the guardian

```javascript
console.log(
  "\n%cHello.\n\n%cWe are hiring – ever thought about joining us? \n%chttps://workforus.theguardian.com/careers/digital-development%c \n\n",
  "font-family: Georgia, serif; font-size: 32px; color: #052962",
  "font-family: Georgia, serif; font-size: 16px; color: #767676",
  "font-family: Helvetica Neue, sans-serif; font-size: 11px; text-decoration: underline; line-height: 1.2rem; color: #767676",
  ""
)
```
Looks like there are styles for each line, interesting. I didn't know these things about console.log
Here's a resource on the matter: http://www.softwareishard.com/blog/firebug/firebug-tip-styled-logging/

ezekielnizamani, 2020-11-08:
code is ready but can't add it to the project cause got some issues here is the code
 
`console.log(
  "\n%cHey, 👋.\n%c______________________________________________________________________________\n\n%cCadHub is in active development - Want to lend a hand? \n\n%chttps://github.com/Irev-Dev/cadhub",
  "font-family: Georgia, serif; font-weight:bold; line-height: 2rem; font-size: 32px; color: #3c366b",
  "font-size: 10px; color:#D3D3D3",
 "font-family: Georgia, serif; font-size: 16px;line-height:3rem ",
  "font-family: Helvetica Neue, sans-serif; font-size: 16px; line-height: 1.5rem; color:#gray",
  
)`

Irev-Dev, 2020-11-08:
Thanks @ezekielnizamani 
Do you want to try and work through the issues or do you want to me to add the code?
It's fine for me to add the code but it means you will miss out on raising a pull request and being listed as a contributor, which is fine if you're not worried about that.

What were the issues that you hit?

PR #109 (2020-11-18): Add welcome console log
resolves #65 Thanks @ezekielnizamani 
[image]
closed 2020-11-18 by Irev-Dev