Irev-Dev/cadhub #81: Make top right avatar into a drop down menu with the following two options: [profile, sign out]

Irev-Dev, 2020-11-08:


Irev-Dev, 2020-11-09:
Hey @Yash-R, I've assigned this one to you. I think it will be good because it's adding a bit of new functionality, and there's an example of something similar being used with the popover. Though another good issue might be #83. Let me know if you want to that instead or neither.

Yash-R, 2020-11-09:
okay, I can do both let do this one first is that okay @Irev-Dev 
And please give me all the details for installing.

Irev-Dev, 2020-11-09:
Great,
And yup thanks for the reminder, working on that now.

Yash-R, 2020-11-09:
@Irev-Dev how long will it take?

Irev-Dev, 2020-11-09:
Sorry took me a while because I was setting a database seed. not the quickest way to get you up and running but good to have for future.
So you'll need to pull the latest main branch to get the changes I just made. Then you need to delete the sqlite db. it will be located at `api/prisma/dev.db`, so just delete this file.

Then run 
```terminal
yarn rw db up
yarn rw db seed
yarn rw dev
```
sign into the app with

email: localUser1@kurthutten.com

password: abc123

and you should be good to go. Let me know if you have any problems.

Yash-R, 2020-11-09:
thanks its working good now

Yash-R, 2020-11-09:
@Irev-Dev how can i access userName i have tried all posiible solution but the only error i got is userName is not defined

Irev-Dev, 2020-11-09:
@Yash-R,
UserName is: `local-user-1` and 2

http://localhost:8910/u/local-user-1

You can see the details of what seed data was added in `api/prisma/seeds.js`

Not directly related, but the sqlite browser can be handy for directly checking what's in the db. https://sqlitebrowser.org/dl/ once installed use it to open `api/prisma/dev.db`, just be aware that if you make changes to the data with sqlite brower this button `write changes`
[image]
Will light up, and if you don't click it than it's like the sqlite browser takes control of the db and it can start blocking our code from doing writes and you'll start getting errors out of now where. Can be confusing unless you know the two can conflict.

Yash-R, 2020-11-09:
Okay but I think there is some problem in useAuth because the new button is not working and current User is also not accessible.

Yash-R, 2020-11-09:
The problem is how to access user data in main file.

Irev-Dev, 2020-11-09:
Hmm, okay I'm not sure.
If I add a console log into MainLayout like so
[image]
Then I get an object like this in the console.
[image]
Does that work for you?
I don't know if signing out and back in might help 🤷

Yash-R, 2020-11-10:
[image]
[image]

@Irev-Dev its looking like this do you want me to change anything?

closed 2020-11-10 by Yash-R

Irev-Dev, 2020-11-10:
@Yash-R Looks great. Really glad to see you've got some design chops. Even simple things like a decent amount of padding to text can breath is not something I take for granted. Only tweak I'm thinking of is it might be good to align the right edge of the avatar with the right edge of the popover, what do you think?
I think that would be:
```
anchorOrigin={{
    vertical: 'bottom',
    horizontal: 'right',
  }}
  transformOrigin={{
    vertical: 'top',
    horizontal: 'right',
  }}
```
In the materialUi popover.

PR #90 (2020-11-10): made drop down menu
resolved #81