Irev-Dev/cadhub #102: Add React Helmet

Irev-Dev, 2020-11-17:
Important for SEO reasons,

Plus we have new log. We'll probably go with the small gear-donut with a bit out of it as the gravatar.

Matheusdoe-dev, 2020-11-17:
Can I get this issue ? I can do this task setting dynamic title and seo meta tags on pages.

Irev-Dev, 2020-11-17:
Awesome thanks @Matheusdoe-dev .
Let's make it so that when someone is on a user page eg `/u/local-user-1`
We'll update the title to their username and update the social media image to their avatar.

and when someone is on a part page eg `u/local-user-1/demo-part1` or `u/local-user-1/demo-part1/ide`
We'll update the title to the part title and the social media image to the part's main image.

How's that sound?

Irev-Dev, 2020-11-17:
Oh and since you've worked on this project before, there's been an update to the submodules so you'll need to pull recursively.
`git pull --recurse-submodules`
and you need to run `yarn move-cad-worker`
and the db has changed, easiest thing to do will be to delete and reinit
Delete the file at `api/prisma/dev.db`
Then run 
```
yarn rw db up
yarn rw db seed
```

If you run into issues, probably the easiest thing is to start with a fresh clone and follow the instructions in the readme.

Matheusdoe-dev, 2020-11-17:
Right, I will use git clone again, is better.

Irev-Dev, 2020-11-18:
resolved by #113

closed 2020-11-18 by Irev-Dev