Irev-Dev/cadhub #104: Add basic project search

Irev-Dev, 2020-11-17:
Something basic will do using native database quries will be fine. See the prisma docs
https://www.prisma.io/docs/concepts/components/prisma-client/filtering#filter-conditions

We should be able to use "contains" on the title and description for a very basic search.

We might add elastic search or algolia later for a more advanced implementation

if you want to pick this one up, let me know and I'll try and get you designs soon. But it's basically going to be a search field, then displaying projects with the projects cards that already exist on the home page https://cadhub.xyz/

bhargavkumar7031, 2021-05-11:
Hi @Irev-Dev I would like to work on this

Isaac-Tait, 2021-12-10:
Came over after your RedwoodJS talk the other day. Would Algolia be a good solution for this? I set it up on my GatsbyJS site and their free tier works well...

Irev-Dev, 2021-12-11:
Hey @Isaac-Tait, Oh nice, presenting is paying off 😁 

I think Algolia would be perfect.

Just to be clear are you offering to implement this feature? If so should I assign it to you?

Isaac-Tait, 2021-12-11:
Yes please. I would be happy to be assigned this issue. I am in the midst of implementing Algolia on another project. So, this would be a good project for me to take on.

Irev-Dev, 2021-12-12:
Awesome, have at it!

I've had a chat to @franknoirot and he's going to put together some designs for the search. It's up to you if you also want to implement the designs. If you're able to get things set up with Agolia and have a basic version working. I'll be happy to match the designs 😁

Isaac-Tait, 2021-12-13:
Sounds great. I would be happy to implement the @franknoirot designs. I will get started on this asap 🤩

franknoirot, 2021-12-13:
@Isaac-Tait excellent, I'll make some designs and comment them in here after work today. Feel free to tag me with any clarifying questions.

franknoirot, 2021-12-14:
@Isaac-Tait I've made just a few designs within [this page of our Figma project](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/?node-id=2313%3A5666), and I've added some functional and design notes as comments within there. While you're in the file you can hit the `C` key to enter comment mode and you should see them as clickable icons over the designs.

After chatting with @Irev-Dev here are some goals for search divided between this V1 implementation and future work:
- This iteration
  - Fuzzy text search of project titles
  - Filtering by CAD package
  - URL-driven search (if you visit https://cadhub.xyz/s?q=donut&c=openscad it should load with the search and filter applied)
- Future work (just as an FYI and for us to remember)
  - Pagination or virtual list
  - `@` to search for people
  - Sorting of results
  - Grid/list view toggle

Irev-Dev, 2021-12-14:
@Isaac-Tait I only have a crude understand of how solutions like agolia work, from what I understand is you tell it which database columns to index for the search? in which case would the project title and the project description make sense?

Isaac-Tait, 2021-12-16:
I am running into an error right off the bat. I cloned the `cadhub` repo and then ran `yarn install` (according to the [contributing](https://github.com/Irev-Dev/cadhub/blob/main/CONTRIBUTING.md) documents) but when I run `yarn rw prisma migrate dev` I get the error: 
```
error Couldn't find a package.json file in "/Users/isaactait/Desktop/Web_Dev"
```
This error cropped up earlier this year in RedwoodJS but was patched... https://github.com/redwoodjs/redwood/issues/222 At this time I am not sure if the problem has resurfaced or if something is not right on my end (the most likely culprit 😂). 

Can anyone confirm that this is an issue for them as well?

It should be pretty easy to duplicate if you 1️⃣ fork/clone the repo, 2️⃣ cd into it,  3️⃣ run `yarn install` and 4️⃣ then `yarn rw prisma migrate dev`

Irev-Dev, 2021-12-16:
Did you `cd` into the `app` directory @Isaac-Tait?

Yarn install and other commands won't work from the root directory of the repo.

[image]

Irev-Dev, 2021-12-16:
maybe I should update the docs to `cd cadhub/app`

Isaac-Tait, 2021-12-16:
Yup, I was pretty sure it was something I was doing wrong on my end. I understood the docs to mean `cd into cadhub` not drill down one more level. Once I went into `app` it went smoothly. Thank you for getting me going in the right direction 🤣 What you have now in the docs makes sense now that you pointed it out. Perhaps updating the docs would make it clearer... or I could have just paid better attention 🤷🏻‍♂️

Irev-Dev, 2021-12-16:
Done 😁 

https://github.com/Irev-Dev/cadhub/commit/cd3060b3c729d791f8965c93f939304067dd91b8

Isaac-Tait, 2021-12-18:
I am figuring this out for the first time as I go and I am stuck on how to proceed through connecting to a local DB.

I am at the point where I need to [connect to Postgres](https://redwoodjs.com/docs/local-postgres-setup#connect-to-postgres). It says I should 

> Add a DATABASE_URL to your .env file with the URL of the database you'd like to use locally.

If I follow the Redwood "tutorial" and use _their_ database URL I get an error saying I do not have permission to access the database. 

If I run `psql` I see that I have four databases. So, how do I get the URL to one of those databases to use locally?

Isaac-Tait, 2021-12-18:
> Generate a new page with `yarn rw g page search` and use the TopNav to make the search bar, that component takes children to go in the center of the bar, so just put the search as a child.

Thank you for the explanation. Makes sense, I should be able to do that 😄  The figma designs are very cool, thank you for those @franknoirot 

> I only have a crude understand of how solutions like agolia work, from what I understand is you tell it which database columns to index for the search? in which case would the project title and the project description make sense?

I can have Algolia populate an index via a GraphQL query. I can also setup multiple indices from two queries too (a feature I am excited to try out as I have read about it but have yet had an opportunity to implement). So, a search for designs and @people would come from two separate indices that are created in the Algolia "backend" 

If I am understanding your question then yes searching for a cadhub project title and/or cadhub description would be the best way to go.

Irev-Dev, 2021-12-18:
@Isaac-Tait you env var should look like the one in the redwood docs `DATABASE_URL="postgresql://postgres@localhost:5432/redwoodblog_dev?connection_limit=1"` where `postgres` is the username and `redwoodblog_dev` is the db name, the latter you can change to a sensible name, maybe `cadhub_dev`, that becomes the name of the db. postgres will create the db iff it doesn't exist.

If you have permission issues, they are permission issues with postgres itself and the user, not a particular db. It would be useful if you post the error you are seeing but to take a swing. What user are you using?

When I type `psql` it shows me my user / whoami is `kurthutten`
[image]
Therefore my url is `DATABASE_URL="postgresql://kurthutten@localhost:5432/cadhub?connection_limit=1"`

If you are still having permission error I think you might need to check the permissions off your user.

What OS are you using?

Hope that helps.

Isaac-Tait, 2021-12-29:
@Irev-Dev Great, that cleared up the error I had encountered. I am up and running now 🥳

Irev-Dev, 2022-01-07:
Good news! How's it been going?

Irev-Dev, 2022-01-13:
Sorry I'm a little slow to this.

I appreciate you pushing through this. Interesting the use of GraphQL is different.

How can I help? If there's a chunk of work, or something you'd like me to investigate?

In regards to the you not having any projects locally I can probably figure out solution to that.

Isaac-Tait, 2022-01-16:
> In regards to the you not having any projects locally I can probably figure out solution to that.

It would be awesome if you could help with that. Thank you 👍🏼 😄 

Right now I created a [search page](https://github.com/Isaac-Tait/algolia-cadhub-search-project/blob/0ede72739d53d1ab05f5772841d8a54696133cd5/app/web/src/pages/SearchPage/SearchPage.tsx). The next step is to sort out the graphQL query that will send the CadHub projects to the Algolia "Design" indices. Algolia will convert that into JSON data that will be used to fulfill searches from the client... 

Once we get the search page up and running it should be fairly straightforward in getting the Figma designs up and running. 

Gatsby has a pretty cool graphQL interface called [GraphiQL](https://www.gatsbyjs.com/docs/how-to/querying-data/running-queries-with-graphiql/). The "Explorer" pane makes it pretty easy to build and test queries before pushing them to your site. It also makes it pretty easy to use GraphQL without knowing much about GraphQL 😆

Isaac-Tait, 2022-01-16:
The nice thing about utilizing a GraphQL to populate the Algolia indices is that every time you build and deploy the site the Projects are sent to Algolia. So if you add a new project or update the information within a project then build and deploy the site Algolia will get the new updated information and instantly provide that to the client during a search. That does raise a question though. When a user uploads a new project does that trigger a build/deploy within Netlify? If it does not then you would have to build/deploy the site to trigger Algolia to crawl the projects and update the indices. That seems like it would be a huge pain so if that is the case then I might need to look into how to solve that...

Irev-Dev, 2022-01-17:
Okay I'll look into getting more projects locally, I'll try and see if I can solve that by the end of the week.

Redwood boots up a graphql playground with the dev process (very similar to graphiql) I think it's at http://localhost:8911/graphql

Short answer to rebuilding and redeploying when users add or change a project. No. We simply can't do that, I heard of people using Gatsby when they shouldn't have for pages with user content and regretted for a similar reason, it required rebuilds all the time and the builds would get slower and slower over time.

I haven't used Agolia before, but have worked adjacent to elastic search, and I know we leveraged a feature that allowed the index it be added to over time with individual data/projects, so initially you do a full index of the site, but after that you give it a single project at a time as users add or change them. Maybe Agolia has a similar feature?

Isaac-Tait, 2022-01-20:
Okay, I had an idea in the middle of the night last night. Instead of trying to push the data to Algolia using GraphQL maybe we could just manually upload it? The data would just need to be in JSON format for me to add it to Algolia... Once that is done and we have search functionality we can figure out how to update regularly. This way we are breaking it down into smaller chunks... What format is the db data currently in and can you send it to me (Dropbox, GDrive...)?

Irev-Dev, 2022-01-22:
The db is a postgress db, and I don't think it's a good idea that I send it to you. However I have found a solution for local projects. is isaac@mountaintopcoding.com the best way to contact you @Isaac-Tait?

Isaac-Tait, 2022-01-24:
@Irev-Dev yes that is my personal email and the best way to get a hold of me. Hope you had a nice weekend 😄

Irev-Dev, 2022-01-24:
Sweet thanks,

In regards to the index issue does this help https://www.algolia.com/doc/guides/sending-and-managing-data/send-and-update-your-data/in-depth/the-different-synchronization-strategies/#full-record-updates
It sounds like full-record-updates would suit us no? as we know records change, we can update the record as apart of creating or updating project?

Isaac-Tait, 2022-02-11:
I [pinged](https://community.redwoodjs.com/t/question-about-cells-in-regards-to-algolia-search-functionality/2755/2) the Redwood forum regarding this project and got some very helpful feedback from Rob and David T. While sorting out how to write a script to populate the Algolia indices I came across [Algolia Crawler](https://www.algolia.com/doc/tools/crawler/getting-started/overview/). It works with the free plan with sites hosted by Netlify, which CadHub is. 

The docs state 

> The Crawler is an automated web scraping program. When given a set of start URLs, it visits and extracts content from those pages. It then visits URLs these pages link to, and the process repeats itself for all linked pages. With little configuration the Crawler can populate and maintain Algolia indices for you by periodically extracting content from your web pages.

You would need to create an Algolia account (something you need to do regardless of how we go about setting this up) and then link it to your Netlify account by following [these](https://www.algolia.com/doc/tools/crawler/netlify-plugin/quick-start/) guidelines. 

In my cloned CadHub repo I got search running on the [homepage](https://github.com/Isaac-Tait/cadhub/blob/main/app/web/src/layouts/MainLayout/MainLayout.js) and on a separate [search page](https://github.com/Isaac-Tait/cadhub/tree/main/app/web/src/pages/SearchPage). 

[image]
---
[image]

The separate search page is ugly (un-styled) and mainly just a proof of concept... Right now when I search there are no results cataloged in the Algolia indices so no search results are returned, which is where the Crawler would come in. Obviously I do not have access to the Netlify account to link the Algolia Crawler to (I tried using my own Netlify account but it would not allow me to create the Crawler - kinda makes sense). Would you be willing to give it a try setting up the Crawler? I had never heard of it until today, so I am not sure how well it works but on paper it looks really promising.

Cheers!

Irev-Dev, 2022-02-20:
Sorry I've been a little MIA of late, thanks for looking into all this.

I'm not so sure about the crawler though I could probably set it up as an experiment. My concerns are 
- The netlify hosting is likely to change, as I'd like it to change to deploying with the serverless framework on AWS. I couldn't find anywhere that said it has to be netlify with the free teir?
- I'm not sure it's fit for purpose, it does sound easy to set up, but looking at their FAQ
[image]
Our data accessible from it original source, or am I miss-interpreting that?