Irev-Dev/cadhub #176: Profile Pic Does Not Save

jmwright, 2020-12-28:
I started playing around on CadHub.xyz and tried to set my profile picture by selecting the image and clicking the "Upload" button. A green flash message shows at top saying "User Updated" and there are no errors in the web console, but when I leave and come back I see that my profile picture wasn't saved.

Version: Current Pre-Alpha on cadhub.xyz
Browser: Firefox 84.0 (64-bit)
OS: Ubuntu 18.04 64-bit

jmwright, 2020-12-28:
It must have just taken a bit to update or something. I closed the tab and reopened just now and my profile pic is there. Sorry for the noise.

closed 2020-12-28 by jmwright

Irev-Dev, 2020-12-28:
Thanks @jmwright. 
I'll take a look and try and get this to update straight away.