Irev-Dev/cadhub #239: Adjust openscad image to match viewer size

Irev-Dev, 2021-03-12:
`web/src/helpers/cadPackages/openScadController.js` is hardcoded to get a 500x500 imaged from the openscad renderer. This should instead match the size of the viewer panel. I would assume that `react-mosaic` has some kind of event for resizes that we can fire to adjust it.

It should also detect if the device is retina display and double the resolution.
https://stackoverflow.com/questions/19689715/what-is-the-best-way-to-detect-retina-support-on-a-device-using-javascript#20413768

franknoirot, 2021-03-14:
Hey @irevdev#1888 in `openScadController.js`, is the `size` property supposed to control the size of the image returned from OpenSCAD? I've been able to get the image full-size and added in the `viewerSize` based on the on-screen size of the element to `IdeState`, passing it into the render call, but I'm still getting a 512 x 512px image back from OpenSCAD.

Irev-Dev, 2021-03-14:
I'm guessing 512 is the default for openscad, I've had a quick look, and the number look like floating points to me not integers.  so I'm guessing it's falling back to 512. Try rounding the numbers to integers.

I'll back to this in a couple hours. I should be able to tell you something more definite. ^ that's my suspicion.

franknoirot, 2021-03-14:
Gold, that did the trick. I'll push up a commit now and then get the retina thing sorted in a final one before PR.

franknoirot, 2021-03-14:
@Irev-Dev okay so now I'm on another issue, seems related to a comment you left on line 118 of `IdeViewer.js` which I've tried to demonstrate in my latest commit. When the panes are dragged the image now corrects by fetching a new image ✔, but after having dragged the pane if you try to orbit the camera the preview uses a stale `viewerSize` value and goes back to incorrect proportions. You mentioned in that comment the need to use a locally-defined variable to overcome the stale code issue within this file, but though I've been able to do the same with `viewerSize` within the component itself, what is interesting to me is that the `render` dispatched event within `ideViewer` uses a stale `viewerSize` value even when that value is not passed into the `render` call, but used within `useIde`. I'll investigate more in the morning just wanted you to see what I'm working through, because it might be something a bit larger to sort out now before this stale state thing gives us perennial headaches.

Irev-Dev, 2021-03-14:
@franknoirot  I had a bit of a play earlier and got something working.

One thing I did is to manage the viewer dimension state completely in the openscad controller so that openscad specific state isn't dirtying the ide state, it make other things more complicated, in that at the start it's expecting both the ideContainer and the viewer to call render, one gives it it's dimensions and the other it's camera and it doesn't return an image until the second render call, but other than that it also means it automatically renders a new image on resize.

So I think I've likely side stepped the issue you've run into. but I had ran into stale state in ideViewer too (I think somehow `react-three-fiber` not being completely react through-and-through might be causing it) I got around it by raising a variable to module scope and updating it each react render. https://github.com/Irev-Dev/cadhub/blob/main/web/src/components/IdeViewer/IdeViewer.js#L118

If my work is of use to you you can checkout `kurt/239` or look at the [commit](https://github.com/Irev-Dev/cadhub/commit/b6867fc8a81f01878c1cd0fd0cfde9aa6e41db36?branch=b6867fc8a81f01878c1cd0fd0cfde9aa6e41db36&diff=split)

Irev-Dev, 2021-03-14:
I update the above comment a couple times after re reading your comment.

franknoirot, 2021-03-15:
Ah nice, I thought the `viewerSize` would be useful across CAD packages initially but I like what you did way better, that's working great. I think you got it if you want to PR, sorry I couldn't be of more use on this one. Love the new sprinkles lol

Irev-Dev, 2021-03-15:
okay, I've put up #251, sorry didn't mean to steal your thunder.

franknoirot, 2021-03-17:
No no worries!

franknoirot, 2021-03-17:
@Irev-Dev okay I'm messaging in here before a PR because it seems to simple to be correct. I know you linked to that StackOverflow discussion but what I've put in [my commit](https://github.com/Irev-Dev/cadhub/tree/franknoirot/239) is inspired by [this MDN article](https://developer.mozilla.org/en-US/docs/Web/API/Window/devicePixelRatio) about using the `Window.devicePixelRatio` property to correct the resolution of a canvas, which I all I believe we need correct? Just generate an OpenSCAD render that is the size of the container multiplied by the pixel density? It seems to work on my laptop, which has a pixel ratio of 1.25. Let me know if I'm misunderstanding the problem.

Irev-Dev, 2021-03-17:
@franknoirot Sounds like you understand the problem better than me 👏

closed 2021-03-19 by Irev-Dev