Irev-Dev/cadhub #240: Convert openscad renders to jpeg or webp

Irev-Dev, 2021-03-12:
Images are being sent as png which is not the best web format. jpg should be much smaller, and if we can detect if they're browser supports webp that would be even better.
https://stackoverflow.com/questions/5573096/detecting-webp-support#27232658

We'll need to figure a good way of doing this in node.

There's a possibility that compressing the image then sending will take longer than just sending, but I doubt it 🤷

toddmedema, 2022-01-13:
I noticed that it has to re-render each time you rotate the camera - why are the renders being sent as an image? Couldn't they be sent as a 3D model so that you can play around with them in the browser directly via web.gl? e.g. https://threejs.org/examples/webgl_loader_stl.html

Irev-Dev, 2022-01-14:
This is why.

https://learn.cadhub.xyz/docs/general-cadhub/openscad-previews