Irev-Dev/cadhub #319: ImplicitCAD support

julialongtin, 2021-05-16:
Implicitcad is another scad engine, with different properties, and an engine written in haskell. see: github.com/colah/ImplicitCAD

Irev-Dev, 2021-05-16:
Hey @julialongtin,

Definitely familiar with ImplicitCAD and definitely interested. Because you folks already have an online editor I've had a bit of a poke around previously. It seems very performant, must be the Haskell :)

Right now I'm trying to balance adding integrations and building out other lacking features on CadHub.

If you (or someone) is willing to help out with the integration then I'm definitely willing to seize that opportunity. So by help I mean with the backend/3d generation. I'll handle all the web app stuff? How does that sound?

From having poked around with your current editor, it looks like what is returned is an entire js script that gets executed? Do you know if we can download a 3d format instead, `.gltf` perhaps?

Happy to talk more about the how we setup the other integrations (long story short is we're using aws docker-lambdas to generate the 3d files and dumping them in an s3 bucket).

Irev-Dev, 2021-05-17:
I've checked the readme, not `.gltf`, but `obj` or `stl` should work fine :)

julialongtin, 2021-05-17:
hey, i'd like to have a more extended conversation. would you mind either popping into irc (#implicitcad on freenode), or setting up a call?

Irev-Dev, 2021-05-17:
That sounds great.

I've popped into the irc, I'll endeavour to hangout there, but I think a call would be good to touch base.
Figuring out a time to call might be easier on IRC, but I'm in AEST (UTC +10) but am up early so can normally jump on a call anytime after 5am my time.

In terms of what to call on,
I'm normally [on discord](https://discord.gg/SD7zFRNjGH)
But I've also had good experiences using [jitsi](https://jitsi.org/)

Do you have a preference or suggestion?

Rylangrayston, 2022-03-26:
I like ImplicitCad as well and would choose it for some designs on CadHub