Irev-Dev/cadhub #320: Add OpenSCAD Customizer parser

Irev-Dev, 2021-05-16:
We want to offer a customizer for the parts on OpenSCAD. So the first part of that is simply being able to parse openscad code to pull out the customizable params.

Customisable params look like this in openscad
```
// Text box for string
String="hello";
```
Which becomes 
[image]

There's lots that will need to go into the customizer but this ticket is to generate a data structure from parsing the Openscad code something like:
```
[
  {
    parameter: 'String',
    label: 'Tex box for string',
    type: 'string',
    defaultValue: 'hello",
  },
]
```
but more elaborate for all the other types.

This is a good ticket for new contributors to CadHub as it doesn't require knowledge about our stack, it's purely input output and can be developed with tests to make sure it's correct.

Where to find examples for the tests? use the documentation at https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer

Irev-Dev, 2021-05-16:
Side note is that Kintel on the core team of OpenSCAD had share this repo of a customizer built with Angular that might be useful.
https://bitbucket.org/shapefactory/scad-web/src/master/

Having looked at it myself I found it pretty difficult to read so not sure how useful it might be, I at least found lines like this one that are looking for the start of comments `//` so probably related to parsing params.
https://bitbucket.org/shapefactory/scad-web/src/6f1b3c4ebf73d012ec3117d4f488a5f383aa19e5/public/js/design.js#lines-92
it's also from 2015 so I'm not sure if the customizer has changed since then.

t-paul, 2021-05-16:
OpenSCAD already has code to save parameter sets as JSON file which can be applied using `-P / -p` command line arguments. Code for an export format to write out the data for the script values never made it to `master` but should be easy to add.
Benefit would be that it could seamlessly integrate with a potential native customizer syntax based on some sort of [annotations](https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases) in code instead of parsed comments.

Irev-Dev, 2021-05-16:
I think I follow what you're saying @t-paul but to be clear.

What I'm trying to get is some kind of data structure that we can easily convert to UI ( so metadata about if it's a slider, text input etc, it's default value and so on) and I'm think of doing that on the frontend with a js parser. so are you saying that maybe we could output something like this directly from OpenSCAD instead of us trying to parse code?

Because that sounds like a really good idea, it makes sense for OpenSCAD to be the source of truth for this stuff instead of us potentially maintaining our own parser too.

t-paul, 2021-05-16:
Yes, the current datasets only have the parameters and values that are selectable by name from the customizer UI. For example: https://github.com/openscad/openscad/blob/master/examples/Parametric/sign.json

A similar structure with meta data and default values could be exported to stdout by running `openscad -o - --export-format param file.scad`. Pretty much as shown in the initial post.

I would go more into direction of exporting the type information and leave it to the UI to decide for a control to use. In OpenSCAD the customizer is currently relatively fixed regarding what control it shows (e.g. integer + a given value range will produce a slider). For now it's fine as only the Thingiverse feature set is supported and there's no plan to make incompatible changes or extend the comment parsing.

But ideally, with more powerful native annotations, I think that should be just a hint and it would be nice for the user to be able to select a different UI control if that makes things easier, e.g. changing the slider to a spin-box which allows typing values.

Irev-Dev, 2021-05-17:
That sounds fantastic @t-paul,

Are you confident you'll be able to get into OpenSCAD? cause I'll change the title of this to reflect integrating with the this feature instead of parsing ourselves.

> I would go more into direction of exporting the type information and leave it to the UI to decide for a control to use.

Yup I agree with that too :)

Irev-Dev, 2021-06-06:
I've edited the original description to be inline with the discussion here.

Irev-Dev, 2021-08-22:
Related: https://github.com/openscad/openscad/pull/3864