Irev-Dev/cadhub #365: Theme OpenSCAD preview image with CadHub colors.

Irev-Dev, 2021-06-15:
@franknoirot dug up this gem
http://forum.openscad.org/Setting-transparent-background-td18272.html

and I've experimented and got something working locally the following is the DeepOcean theme with a bluer background
[image]

The background is the primary concern, but we could change all the colors if we wanted.

I'll make a PR when we have colors in mind, now I'll just list the changes needed to be made here.

1) Make a `app/api/src/docker/openscad/cadhubtheme.json` file
```json
{
  "name" : "CadHub",
  "index" : 1600,
  "show-in-gui" : true,

  "colors" : {
      "background" :         "#333355",
      "axes-color" :         "#c1c1c1",
      "opencsg-face-front" : "#eeeeee",
      "opencsg-face-back" :  "#0babc8",
      "cgal-face-front" :    "#eeeeee",
      "cgal-face-back" :     "#0babc8",
      "cgal-face-2d" :       "#9370db",
      "cgal-edge-front" :    "#0000ff",
      "cgal-edge-back" :     "#0000ff",
      "cgal-edge-2d" :       "#ff00ff",
      "crosshair" :          "#f0f0f0"
  }
}
```

2) add the following line to `app/api/src/docker/openscad/Dockerfile`
`COPY openscad/cadhubtheme.json /usr/share/openscad/color-schemes/render/`

3) Change `--colorscheme DeepOcean` to `--colorscheme CadHub` in `app/api/src/docker/openscad/runScad.js`
note the name is base of the name in the JSON not the name of the JSON file.

Irev-Dev, 2021-06-15:
Theming for the editor shouldn't be too hard for two reasons, 1) you can base it of another theme and modify it, 2) you can use their dev tools by hitting `F1` then pulling up `Developer: Inspect Tokens` to figure out what tokens you want to change.

Here's some stuff I was playing with at https://microsoft.github.io/monaco-editor/playground.html#customizing-the-appearence-exposed-colors
```
monaco.editor.defineTheme('myTheme', {
    base: 'vs-dark',
    inherit: true,
    rules: [
        { background: '222222' },
        { token: 'comment', foreground: '5098F1', fontStyle: 'italic' },
        { token: 'keyword', foreground: '8732F2'},
        { token: 'string', foreground: 'DF5CA0'},
    ],
    colors: {
        'editor.background': '#1A1A1D',
    }
});
```
[image]

franknoirot, 2021-06-16:
Love it! I don't know why I can't get your code to work in there; whatever I did appears to not recognize the Python comment syntax for some reason.

franknoirot, 2021-06-16:
False alarm I got myself sorted out, I had left the `language: "text/plain"` in there by accident.

franknoirot, 2021-06-16:
Alright @Irev-Dev I just tweaked the colors a bit from those top nav colors because they were vibrating a bit to my eyes when directly on that dark background, and added a `foreground` attribute to get the default text color matching. Does that look alright to you?
```javascript
monaco.editor.defineTheme('myTheme', {
    base: 'vs-dark',
    inherit: true,
    rules: [
        { background: '222222' },
        { foreground: 'CFCFD8'},
        { token: 'comment', foreground: '79B2F8', fontStyle: 'italic' },
        { token: 'keyword', foreground: 'A663FA'},
        { token: 'string', foreground: 'F98CC5'},
    ],
    colors: {
        'editor.background': '#1A1A1D',
    }
});
```
[image]

Irev-Dev, 2021-06-16:
@franknoirot  Yup looks good. I'll get a PR up soon

commit 142bf2c8 (2021-06-16): Start adding ide colors, includes adding custom theme to openscad

Resolves #365
PR #366 (2021-06-17): Start adding ide colors, includes adding custom theme to openscad
Started to add colors but didn't go too crazy yet.
I was thinking @franknoirot is it worth adding the colors as "elements" in figma (not sure what the correct jargon is) but have the names match up between class names and names in figma?

Here's a screen shot.
[image]

This gap is going to be difficult to remove, as if I override the mosaic styles to remove the margin, than there's no gap between panels.
[image]

I decided against the save icon for when the button is labeled "preview" is that okay?
[image]

Resolves #365
closed 2021-06-17 by Irev-Dev