Irev-Dev/cadhub #383: CQ models rendering not usable

adam-urbanczyk, 2021-06-26:
Thanks for all the work on integrating CQ. Sadly, current rendering of models is not really usable - I cannot really see any features/edges:

[image]

What would you need from CQ to fix it? Currently VRML and VTKJS (take a look here: https://kitware.github.io/vtk-js/examples/SceneExplorer/index.html?fileURL=https://data.kitware.com/api/v1/file/5878eb978d777f05f44b142a/download)  export is supported, it should be easy enough to display those with descent shading. Or would another format be needed?

Irev-Dev, 2021-06-26:
Thanks @adam-urbanczyk 

This is just bad lighting in the three.js scene. No changes in format is needed.

Irev-Dev, 2021-06-26:
I've improve it somewhat with #384 
[image]

But I'll leave this issue open as I'd like to improve it further.

marcus7070, 2021-06-27:
You might be interested in this: https://github.com/CadQuery/sphinxcadquery/blob/0b7050c0204b8b2c6f7e6912eaee97ca02b9e776/sphinxcadquery/sphinxcadquerystatic/main.js#L130-L165

I think you can see that lighting in action here: https://paramak.readthedocs.io/en/main/example_parametric_reactors.html

Irev-Dev, 2021-08-10:
Once #448 is merged, I was planning on closing this issue, while I hope it continues to improve I think it's much better than it was. Surfaces and edges can easily be identified now.

https://user-images.githubusercontent.com/29681384/128854453-a6960d50-cf06-4cd6-ad83-803acb96855e.mov

Are you happy with the state of it now @adam-urbanczyk @marcus7070 ?

closed 2021-08-10 by Irev-Dev

adam-urbanczyk, 2021-08-10:
LGTM, I'd change the line color though (to eg black)

marcus7070, 2021-08-11:
It's a great improvement @Irev-Dev. I think your line colours are part of the CadHub theme? In which case they're perfectly legible, no worries.

A minor point: I think most CAD viewers spin the object, not the camera. Sometimes you might get a shadow or reflection that obscures a feature, if you're only able to move the camera then you can't un-obscure. On the other hand, I do think spinning the camera with fixed lights gives a better sense of depth to the scene, but maybe that's just me.

Irev-Dev, 2021-08-11:
Thanks @adam-urbanczyk, @marcus7070,

Why do you think the lines should be black Adam? The lines are roughly a CadHub theme, but at somepoint I'll probably want to re think the whole design of the 3D viewer so it's likely to change.

Thats a good point Marcus, I might do a bit of both, I tried attaching a light to the camera, and it adds a fair amount of consistency to the lighting.

adam-urbanczyk, 2021-08-11:
I can barely see them with the current color scheme (black was just a contrasting color - maybe not so good given the background).

PR #448 (2021-08-10): Add gizmo and edges
Add edges to the models with helps solve the issue of surfaces blending together and should resolve #383 

Also added a gizmo, there seems to be an issue with the animation/camera path, not sure if its my setup or a bug (asked here https://github.com/pmndrs/drei/issues/465#issuecomment-895812677) but I think its worth adding anyway since at the end of the day it still works for snapping the camera to axis.

@franknoirot something to keep in mind, I never gave a great deal of thought to the pink color of the models, we've recently added a grid, and this adds edges and the co [... cut]