Irev-Dev/cadhub #411: JSCAD integration

Irev-Dev, 2021-07-17:
I've been talking to  about the a possible JSCAD integration. I'm going to try and describe our architecture below so hopefully I can put into the readme as well.

------

Visually our the setup looks like this
[image]
Because we want to have a lot of consistency from one package to the other and to make integrations easier, theoretically when integrating a new package, the only new things that need to be added is the Cad Kernel. So long as it take code and produce a 3d artifact as well as can logs or error messages it should work fine.

Existing packages (OpenSCAD, and CadQuery) make sever requests to get the 3d artifacts, but this is not a requirement, however we at least want to have the ability to run things outside the browser for future features (such as github integration producing new artifacts on push to master).

New cad kernals should be added into `app/web/src/helpers/cadPackages/` using the naming convention `<kernalName>Controller.ts`. It should have a default export of an object with a `render` function that produces the artifact

hrgdavor, 2021-07-18:
jscad definitely works on server-side, so when the time comes for that part of integration it should not be a problem.

Irev-Dev, 2021-07-25:
I finally got around to adding a branch for you @hrgdavor, it's at `kurt/411-demo-branch`
and the diff is here https://github.com/Irev-Dev/cadhub/compare/kurt/411-demo-branch?expand=1

Basically all I've done is made it so you can navigate to http://localhost:8910/dev-ide/jscad
and but since I've only implemented a dumby jscad controller it will only ever look like this.
[image]
But as soon as your able to return geometry from that controller it should work 🤞 (I'm sure there's something I'm forgetting)

Is that helpful?

`app/web/src/helpers/cadPackages/jsCadController.ts` is a typescript file, but as we spoke about earlier feel free to strip out the types and work is js, the function basically just needs to return an object of this shape for it to work (I think):
```
{
  status: 'healthy'
  message: {
    type: 'message'
    message: 'string'
    time: new Date()
  }
  objectData: {
    data: threejsGeoBuffer
    type:  'geometry'
  }
}
```

Irev-Dev, 2021-09-12:
Resolved by #506

closed 2021-09-12 by Irev-Dev