Irev-Dev/cadhub #435: Safari styling errors

Irev-Dev, 2021-08-03:
I've largely neglected safari and the result is there are a bunch of styling problems 
[image]
Z issues for the login modal

[image]
lots of flex gaps ignored by the browser and the page leaves a gap at the bottom of the page.

[image]
More ignored gaps.

found by @revolter.

Irev-Dev, 2021-08-03:
yup so safari just ignores flex gaps until this year, but we probably need to refactor using padding to support older safari.
https://css-tricks.com/safari-14-1-adds-support-for-flexbox-gaps/

Irev-Dev, 2021-08-07:
Because this is all style related it would be a good first issue for someone, especially if you've used tailwind before.

Irev-Dev, 2021-08-09:
Wait this is a terrible issue for you right @franknoirot? don't you run Windows?

franknoirot, 2021-08-09:
@Irev-Dev I've got a MacBook for work so I think I can do it alright!

Irev-Dev, 2021-08-09:
Perfect.

franknoirot, 2021-08-10:
@Irev-Dev confirmed I've been able to get my dev environment in order and use Redwood's forwarding dev command to open the site running locally from my work computer in Safari. On the hunt for some solutions, think I understand the `z-index` issue and will investigate the rest tomorrow.

franknoirot, 2021-08-10:
Okay so the problem with the gaps is that the `gap` CSS property is only supported in the absolute latest Safari version, which up until now only supported `gap` or the former `grid-gap` for elements with `display: grid` and no gap-like property for elements with `display: flex'.

Typically you could use an `@supports (gap: 1rem)` statement to detect if a browser supports a CSS property, but you can't in this case because it's a _two condition_ support statement targeting Safari: supporting `gap: 1rem` _in tandem with_ `display: flex`. This has sparked some fascinating discussion on the [W3C CCSWG](https://github.com/w3c/csswg-drafts/issues/3559#issuecomment-621032156) even though this particular iteration of the problem will go away as Safari users upgrade.

In the meantime, there is no CSS-only fix for this solution. As noted in the CSSWG thread, Ahmad Shadeed published a JS solution for detecting support in [this blog post](https://ishadeed.com/article/flexbox-gap/) that explains the detection issue.

Irev-Dev, 2021-08-10:
Nice investigative work @franknoirot  🕵️ ,
I was thinking that we would solve this simply by replacing the gap class currently on the parent with padding/margin on the children, bit of a pain but also very straight forward? 
Or do you think that a more sophisticated solution detecting the problem and adjusting is called for?

Actually would using grid instead of flex work in most of our cases? I don't think we're using flex specific features (like grow/shrink) so maybe that would work?

Irev-Dev, 2021-08-10:
https://twitter.com/fra_michelini/status/1425023649069867009

franknoirot, 2021-08-11:
Ugh @Irev-Dev I ran into a problem tonight. At work we keep our laptops up-to-date with the latest iOS (I'm running 14.5) and you can't downgrade, so I'm not able to recreate the `gap` issue. I'm trying to emulate older versions of Safari using XCode (I downloaded iOS 13.5 Simulator), but you can't use that to simulate desktop Safari, only iPad and other mobile devices.

I have been able to confirm that replacing `gap` TailwindCSS class names with one-sided margins on all but the first/last children is a good replacement that looks the same in other browsers, but I can't recreate the environment to see all of the issues for real right now.

Do you know of a reliable way to emulate old desktop Safari versions?

franknoirot, 2021-08-11:
One good thing I will say is everything looks great on latest Safari haha, even that height issue with the IDE page is gone for me.
[image]

revolter, 2021-08-11:
@franknoirot, but can you reproduce this:

[image]

?

It happens for me in Safari 14.1.2 (16611.3.10.1.3).

franknoirot, 2021-08-11:
@revolter yup I can replace that! I've committed a fix to it in https://github.com/Irev-Dev/cadhub/commit/68346e075ac2b27ba114b671c760dfc7b14112ad

revolter, 2021-08-11:
Oooh, I see now. I don't experience the gap issues either.

Irev-Dev, 2021-08-11:
Only done one so far but its looking promising
[image]
[image]

franknoirot, 2021-08-11:
Nice! Appreciate it.

closed 2021-08-11 by Irev-Dev