Irev-Dev/cadhub #488: font support

JustinSDK, 2021-09-05:
I publish a Emoticon moai. 
https://cadhub.xyz/u/caterpillar/emoticon-moai

It requires the `font` parameter of `text` to vary emoticons. Currently, Cadhub doesn't support the parameter. The `font` support might be scheduled on the further version. :)

Irev-Dev, 2021-09-05:
Hey, sounds good.
Here's another relevant discussion: https://github.com/Irev-Dev/cadhub/discussions/464#discussioncomment-1216768

As @t-paul mention, it wouldn't be hard to give a list of available fonts (in the ubuntu container it's running on), the problem is more of how does the script author indicate that they want a font select UI? Is this supported elsewhere?

t-paul, 2021-09-05:
I've not seen dedicated customizer support for fonts, they just declared this in a pinned post in the thingiverse group:
> Also, all the google fonts from http://www.google.com/fonts have been added to the servers and can be used in text() :)
> https://www.thingiverse.com/groups/openscad/forums/general/topic:1813

There might be a reasonable extension to the customizer markup for declaring an input field shall be a font selector.

Irev-Dev, 2021-09-05:
@JustinSDK In the mean time I've documented the list of available fonts here: https://learn.cadhub.xyz/docs/general-cadhub/openscad-fonts
and made a Cadhub project demoing them here: https://cadhub.xyz/u/irevdev/all-scad-fonts
If that's useful you could link to either of those in your projects that use text.
Alternatively, since we support selects, if there's a few fonts that you think are good you could curate them yourself and put them into a select?
```
// font select for left eye
font_l_eye = "Arial Black"; // ["DejaVu Serif","Arial Black", "URW Gothic L"]
```
[image]

@t-paul I should be able to work with what ever is decided if that's something you want to add. Since the format for string params is already
```
{
      "caption": "Text box for string with length 8",
      "group": "Parameters",
      "initial": "length",
      "maxLength": 8,
      "name": "stringMaxLength",
      "type": "string"
}
```
You could keep the type as string since that's what it is fundamentally and add `input: "font"` or `font-select` in? That way it's backwards compatible, just defaults back to a regular string input if the select isn't supported?

Don't know what the syntax might be? `// [::font-select]` 🤷 ?

JustinSDK, 2021-09-05:
@Irev-Dev 

It seems that font setting is ok when using customizer.

[image]

but not working in the editor.

[image]

Irev-Dev, 2021-09-06:
ahh @JustinSDK, I think thats just because the customizer always overrules with its values, most likely even when its shut. I think if you refreshed the page it would then respect the script, but thats a good find.
Not sure of what the best approach is, maybe reset the customizer if the script changes?

hrgdavor, 2021-09-06:
> Not sure of what the best approach is, maybe reset the customizer if the script changes?

I would not like if customizer is reset on script change, but a reset button that reverts customizer values to inital values from the script would be a good compromise (I have this implemented in my own prototype of jscad, and I find it personally extremely useful)

Irev-Dev, 2021-09-26:
Do you have any feed back on this approach to the customizer @JustinSDK ?
https://github.com/Irev-Dev/cadhub/pull/529#issuecomment-927193171