Irev-Dev/cadhub #517: social card not working for jscad model

hrgdavor, 2021-09-18:
https://cadhub.xyz/u/hrgdavor/Cookie-Shape-Surprise

https://cards-dev.twitter.com/validator
[image]

Irev-Dev, 2021-09-18:
The social card are a bit broken (just with how they look), I'll use this issue to fix that as well.

But in regards to what your raising directly in this issue, I appreciate you giving an example. There are a few things to consider so I'll list them.
- Social media card showing up on different website depends on a pre-rendering service, which is because the app is a SPA twitter etc won't wait for js to load and then execute it, so a pre rendering service fills in the gap. However this means it will take **24-48 hours** or things to update as that how long the pre-rendering service caches things for. this is not configureable with the current service.
- Seeing if the image it's self is working can be done for looking for the og:image meta tag in the browser.
[image]
In this case ti looks like it's not working (http://res.cloudinary.com/irevdev/image/upload/c_scale,w_1200/v1/CadHub/u-hrgdavor-slash-p-Cookie-Shape-Surprise)
- I suspect one way this might happen is if you change the project title as that would be used for the original url to the image, I'll verify if that's the case or not and see there's a fix to update the image url.

hrgdavor, 2021-09-18:
Yeas I changed the title :) that probaly caused it 

this one is ok https://cadhub.xyz/u/matiasmiche/jscad-oil-pumpjack