Irev-Dev/cadhub #525: Edit bio unavailable if bio doesn't already exist

Irev-Dev, 2021-09-21:
[image]
pencil of edit state, but if no bio
[image]

At a guess could be related to keyValue component @franknoirot ?

franknoirot, 2021-09-21:
Yeah @Irev-Dev I'm sure it's the display logic for the BioField Component, either in `userProfileConfig` or `userProfile` itself. Want me to take a try at fixing?

franknoirot, 2021-09-23:
@Irev-Dev I've found other things I can break that need better validation. Making a list of to-do's:
- [ ] Make it impossible to set `Name` to nothing
- [ ] Make it impossible to set `userName` to nothing
- [ ] Make it impossible to give yourself an already existing `userName`, or at least their profile slug is at least kept unique if they do pick a colliding name.

Would you like for this to be its own issue, or can this issue grow in scope a bit? I'll put up a PR that solves the bio visibility real quick for now.

Irev-Dev, 2021-09-23:
I've enforced that Name and userName needing to have length #526, userNames needing to be unique should already be enforced ( I check locally just incase and seems fine) which is enforced on a db level since the userName is unique in the schema
https://github.com/Irev-Dev/cadhub/blob/main/app/api/db/schema.prisma#L19

What's still not great is that these problems are not communicated to the user, there is a toast message for when the name/bio etc is updated, but we should have a toast message when it fails too.

I'm happy for it to continue under this issue.

franknoirot, 2021-09-23:
Excellent @Irev-Dev you made such quick work of that wow! Okay great I just gotta look at what those errors are returning but adding toasts shouldn't be too hard.