Irev-Dev/cadhub #527: Create Embed system

franknoirot, 2021-09-23:
## Feature Description
Create a code snippet and supporting assets to be able to embed CadHub projects on other web sites/pages.

## Background Research
- CodePen has a [blog walkthrough post](https://blog.codepen.io/documentation/embedded-pens/) for users to explain how their embed editor works. 
- CodeSandbox has an [even more thorough page](https://codesandbox.io/docs/embedding) dedicated to their embed.

## Requirements
Technically the IDE or Details pages, or any page for that matter, can be embedded elsewhere via an `iframe`. But the idea is to create an endpoint that displays the model with the optional addition of either the code or the project info in as small a package size as possible. In the future there should be URL parameters that allow for additional configurations. My initial thoughts are:
1. Remove Monaco, React, any Redwood dependencies. Keeping the "editor" styles might mean recreating them.
2. have STL pre-generated so that code evaluation can be removed
3. Have either vanilla JS or Preact / Alpine / Petite Vue for targeted interactivity, things like switch between tabs for example.
4. Remove any layout change features: code or details on left, viewer on right or full-width.
5. Add control to Share dialog to generate embed snippet, maybe allow user to adjust height of embed in advance with a slider or something?

franknoirot, 2021-09-23:
@Irev-Dev here is my proposed URL structure: `https://cadhub.xyz/u/[username]/[project-name]/embed`

Based on:
- Editor view: `https://cadhub.xyz/u/irevdev/tutorial-hinge/ide`
- Details view: `https://cadhub.xyz/u/irevdev/tutorial-hinge`

franknoirot, 2021-09-23:
Just want to note that after multi-file support there will be a need to be able to define which file's code is visible in the "editor" if included in the embed.

Irev-Dev, 2021-09-23:
Okay wow, this sounds like a lot more work than I was expecting 😅 

I don't think I get why so much effort needs to go into sliming it down? Is it because when you load an iframe you effectively have to download two websites?

I think removing monaco make sense, but replacing react with preact (or other) and than having to roll a new fetching solution in order to remove RW deps feels like we're maintaining a completely seperate project for the sake of embeds. (I'm also not sure about if there is going to be bundling complications since using react is a pretty baked in assumption for RW apps.

Having the STL pre-generated make sense, will depend on work I'm trying to do currently.

URL's seem fine.

I guess I'm not completely sure of what the vision for this is? if we use pre-generated STL, why are we including code at all? is this for tutorial's sake? in which case it can't be edited in the embed?, users will need to go to full website if they want to make changes?

I'm just trying to think how I've personally interacted codepen and codesandbox embeds in the past. I guess I'll interact with them and look at the output as part of a blog post, but the moment I want to see the code I'll immediately click through to the pen. Just thinking out loud here, maybe a part viewer embed will be enough? Guess it's going to depend on who embedding. On medium, embeds are small because of enforced width so part viewer only make sense, but someone's blog they could make it as large as they wanted.

hrgdavor, 2021-09-23:
I would agree for embed to be just preview, and maybe also show code with syntax highligh in some way(tabs, or sth).  

To be able to run the code user should open cadhub.

This type of embed is very useful, and I would personally be fine with it.

Irev-Dev, 2021-09-23:
I believe Redwood bundles by page so we should get that for free.

yup cool, I think keeping react is a much better move from at leat getting an MVP working, can always optimize from there.