Irev-Dev/cadhub #540: Minimize button for console

hrgdavor, 2021-09-29:
When working with jscad models, console is not vey useful to me, and even if it was, it is not needed all the time. 

I feel it is wasting space that I could use for other stuff(more space for 3d view and customizer).

Irev-Dev, 2021-09-29:
@franknoirot, @hrgdavor,

How about we add a toggle into the [side tray](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=1561%3A4532)?

hrgdavor, 2021-09-29:
I would try, have not done a PR for some time :)

franknoirot, 2021-09-30:
@hrgdavor @Irev-Dev how does this look?

@hrgdavor if you'd like to work on this I feel like you can get it working with a normal checkbox and I can make a `Toggle` component like this one to swap out later.

hrgdavor, 2021-09-30:
Deal

Irev-Dev, 2021-09-30:
Some context:
We're using a library for the IDE panels `react-mosaic-component` , it's used in `app/web/src/components/IdeContainer/IdeContainer.tsx` and the layout is controlled by the `layout` property in the app state in `app/web/src/helpers/hooks/useIdeState.ts`.
I don't think it will be much more than dispatching the event `"setLayout"` with the new layout (i.e. with or without the console).
And than dispatching `"resetLayout"` to put it back.
I might be wrong though.

closed 2021-10-12 by Irev-Dev