Irev-Dev/cadhub #553: [Adding badge](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=1002%3A184) when people start editing code from someone else's project

Irev-Dev, 2021-10-12:
[Adding badge](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=1002%3A184) when people start editing code from someone else's project ([full designs](https://www.figma.com/file/VUh53RdncjZ7NuFYj0RGB9/CadHub?node-id=1002%3A184))
[image]

The fork button is in `app/web/src/components/IdeHeader/IdeHeader.tsx` so that's probably where changes need to go.
the following logic should help determine if there's been changes to the code

`const hasUnsavedChanges = project?.code && state?.code && project?.code !== state?.code`

and there is already a `canEdit` boolean which effectively is where the project belongs to the current user, since we want to  add tell the user to fork when they have made changes but it's not their project we'll use something like

`const showForkMessage = !canEdit && hasUnsavedChanges`

Now that `showForkMessage` can be used to show the little popover below the fork button.
Changes will look something like
[image]
But you'll have to figure out the popover yourself.

Please use Tailwind for styles and the svg is part of [`@heroicons`](https://github.com/tailwindlabs/heroicons#react) called `exclamation-circle`

-----
Follow the instructions in [CONTRIBUTING.md](https://github.com/Irev-Dev/cadhub/blob/main/CONTRIBUTING.md) to get setup, and of course ask for help here or in the [dev-help discord channel](https://discord.gg/Vgmky37UtD).

If you'd like to contribute the CadHub but this issue is taken or you'd prefer something else, have a look at #535 to see if there's something else you would like

Irev-Dev, 2021-10-12:
Go for it @javibodas 😁 

I've assigned it to you. Let me know if you need any help.

Irev-Dev, 2021-10-12:
Some steps I just realised I missed is that you need to sign in locally with one of the demo users

localUser1@kurthutten.com: abc123

Create a project (plus button up the top right)

Then sign in as another user

localUser2@kurthutten.com: abc123

and edit the project in order to cause `showForkMessage` to become `true`.

javibodas, 2021-10-12:
Hi @Irev-Dev if I dont save the project, the variable 'project' in the context contains null value. And If I make some changes without save it, the project loaded from a different user is the saved one. I making something wrong?