Irev-Dev/cadhub #556: Respect user's changes when forking a project

Irev-Dev, 2021-10-13:
We allow users to "fork" each other's projects, problem is that if the users has already made some changes to the code before forking, the changes will be lost. You task is to save those changes when forking. More details as well as where to start in the video.

https://user-images.githubusercontent.com/29681384/137108716-d1cbd017-0da1-43fa-9795-93ff24eee828.mp4

Files of interest are
- `app/web/src/components/IdeHeader/IdeHeader.tsx`
- `app/api/src/graphql/projects.sdl.ts`
- `app/api/src/services/projects/projects.ts`

Follow the instructions in [CONTRIBUTING.md](https://github.com/Irev-Dev/cadhub/blob/main/CONTRIBUTING.md) to get setup, and of course ask for help here or in the [dev-help discord channel](https://discord.gg/Vgmky37UtD).

If you'd like to contribute the CadHub but this issue is taken or you'd prefer something else, have a look at #535 to see if there's something else you would like

lucas-barros, 2021-10-16:
Hi @Irev-Dev, I would like to work on this

Irev-Dev, 2021-10-16:
Welcome back @lucas-barros!

It's all yours!

closed 2021-10-17 by Irev-Dev