Irev-Dev/cadhub #581: Console not wrapping long lines

guidoism, 2021-12-09:
Happened to me using the OpenSCAD editor.
[image]

Irev-Dev, 2021-12-09:
Thanks @guidoism.

Would you be able to provide some more details on how to replicate?

What code you were running? what browser were you using?

guidoism, 2021-12-09:
Safari Version 15.1 (17612.2.9.1.20) on M1 Mac. 
User Agent: "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.1 Safari/605.1.15"
Code: https://gist.github.com/guidoism/07e7e294a32f967c623cda839076f9bb