Irev-Dev/cadhub #586: special characters in comments not working

UBaer21, 2021-12-29:
When using a non alphanumeric character within comments the script is not executed

like  × ⇔ ↦ № ⁴ ℓ  (and many more)
/* × ⇒ ↦ № ⁴ ℓ*/
https://cadhub.xyz/draft/openscad#encoded_script_v2=eJxLLk1K1TDVtObS11I4PF3hUfskhUdtyxQetUxTeNS4BUhP1tIHAP/KD0w=

Irev-Dev, 2022-01-07:
Oh that's interesting.

Do you use characters a lot in comments?

UBaer21, 2022-01-07:
Yes actually i am using a 6 level keyboard layout which gives me direct
access to all these ( https://neo-layout.org/ ).
And I when was trying to use my scripts on cadhub they failed which could
be broken down to the characters in comments as the sole reason at it
seems.
I thought everything in comments is or at least should be ignored by the
code interpreter.
Now it is difficult to know which can be used and which should be avoided,
because some are working just fine like »« but ›‹ are not.