Irev-Dev/cadhub #606: Parameter customizations made through the Viewer are not included in downloaded STL file

franknoirot, 2022-03-07:
## Expected Behavior
When customizing the available parameters in the Viewer pane, especially the one available on a project's details page, I would expect those customized values to take effect on the downloaded file. However, you can only ever receive the file with the parameters described in the code.

## Steps to reproduce
1. Visit [https://cadhub.xyz/u/franknoirot/silverware-drawer-divider](https://cadhub.xyz/u/franknoirot/silverware-drawer-divider).
2. Adjust the `silverware_sections` to `4` from the default `3`.
3. Click Download.

The display of the model should upgrade to the STL view which is rad, but notice that the tray only has 3 sections.

## Additional notes
- It might be nice to provide a toast notification to the user that says "Downloading STL with customized parameters", so that if they tweaked them but actually wanted to download the original they can be reminded to reset those parameters.
- This feels like a part of the UX that could use a lot of tightening: the relationship between the parameters and the download button feels pretty disconnected not just technically but in messaging, because they are. I hadn't really been designing them as the connected set of interactions that they really are. Worth thinking about how to make customizations as clear as possible.

RagingRoosevelt, 2022-04-15:
I ran into this issue on my own file.  Just a data point that it isn't happening just for you.

drf5n, 2022-10-17:
I note that parameter customizations reset whenever I shrink/close the parameter window.   Maybe the ["reload each camera move?"](https://learn.cadhub.xyz/docs/general-cadhub/openscad-previews/) code is starting things from scratch when you switch windows.