Irev-Dev/cadhub #607: Font requests

metasean, 2022-04-23:
Per [Teque5](https://www.printables.com/social/102569-teque5/about)'s [font testing and ranking in the 3d printing realm](https://www.printables.com/model/71231-font-swatches-tested-ranked), it'd be good to see the following fonts added:

- [osifont](https://github.com/hikikomori82/osifont) - apparently conforms to IS0 specification 3O98, which is required in CAD for some European countries
- [Overpass and Overpass Mono](https://overpassfont.org/) - I'm personally working on a project that is most likely going to use Overpass & Overpass Mono
- [Varela Round](https://fonts.google.com/specimen/Varela+Round)
- [Ubuntu](https://design.ubuntu.com/font/)
- [Orbitron](https://fonts.google.com/specimen/Orbitron)

**_Regardless, I'm super stoked to have stumbled across cadhub!_**

angryseptagon, 2022-04-23:
OCR-A and OCR-B are also good aditions in my opinion, they are well known constant thickness monospace fonts that have been in use for decades. I think there are free to use implementations considering their widespread use and history, but I may be wrong.

Irev-Dev, 2022-04-24:
Sounds good.

I think the fonts would need to be installed on the docker image for them to be available.
https://github.com/Irev-Dev/cadhub/blob/main/app/api/src/docker/openscad/Dockerfile

elasticdotventures, 2022-05-01:
I don't understand the nature of this request.  What about a package that simply generates the font from TTF or SVG font? 

Using RUST crate, import any TTF, output as SVG, then openscad can import SVG and extrude
https://docs.rs/ttf-parser/0.15.0/ttf_parser/svg/index.html

Would be nice to be able to access Google Fonts or similar repository (this preview, render font onto surface kind of minutia is the type of place where I think a universal / cross application openSCAD tool browser & font-preview would make sense to me)

t-paul, 2022-05-01:
@elasticdotventures This asks for installing the TTF or OTF font. OpenSCAD can read that natively but it needs to be installed on the local system. All google fonts would abount to more than 1GB, so that would likely need some sort of separation, but it's possible. I believe Thingiverse has done that.