Irev-Dev/cadhub #616: Website Down; graphql error "something went wrong"

ephetic, 2022-12-13:


Irev-Dev, 2023-01-03:
Thanks, I think the db on heroku is down.

Loki44, 2023-03-04:
Came here with the same issue.

lf94, 2023-04-03:
Getting the same issue now also :)

adam2809, 2023-04-19:
Unfortuanately not working for me as well :   ))

yvanblanchard, 2023-05-16:
Same for me.. (Chrome, Firefox, Safari)

shayded-exe, 2023-06-21:
It be great if the error included **any** details and also didn't just block out the whole page.

hrgdavor, 2023-06-21:
Error is not the problem. The problem is that the author @Irev-Dev, looks like it, has abandoned the project :(

Irev-Dev, 2023-06-28:
Definitely is a lot of things to target lol.
Yeah it was an env variable that changes underneath me, it should be up again, sorry it took me ages to dig into this one.

@lf94 is right, folks are welcome to fork, or contribute.

Irev-Dev, 2023-06-28:
I'll leave this open for a bit longer, and close if it looks stable.

DavidPhillipOster, 2023-10-11:
Still down. I got to https://cadhub.xyz from following the link in Round-Anything. I'm on Safari 16.6 in macOS Ventura 13.5.2

weilonge, 2024-05-04:
Tested on Chrome Version `124.0.6367.119` (Official Build) (`arm64`) in macOS `14.4.1 (23E224)`, and it shows "Something went wrong"
[image]

mrluc, 2025-09-07:
2025, site is still down -- M1, sequoia 15.1

[image]